ros-industrial/abb_experimental #17: irb120: collision material yellow should be rgba="1 1 0 1"

Levi-Armstrong, 2015-02-27:


gavanderhoorn, 2015-02-27:
Should it be? :)

culletom, 2015-02-27:
I was surprised to hear this too! Just checked and urdfs in the abb repository are the same, so guess this will have to be changed across the board. Also, abb_orange and yellow are quite similar, so perhaps it could be changed to red?

gavanderhoorn, 2015-02-27:
I'm not sure we actually have a standard colour for collision geometry.

culletom, 2015-02-27:
Yea I imagine not considering its low importance. I'll just correct the rgba values in this repository and submit a PR

gavanderhoorn, 2015-02-27:
Lets just wait for @Levi-Armstrong and see what his rationale is.

Levi-Armstrong, 2015-02-27:
This is also an issue in the abb repository, but the rgba="0 1 1 1" is cyan so the name does not accurately describe the color. I don't have a preference on the color so either the name should be changed to cyan or the rgba should be updated.

gavanderhoorn, 2015-02-27:
If we are going to fix this, perhaps we should create a shared xacro that just defines all ABB colours. This can then be included in all ABB urdfs, making sure colours are consistent.

shaun-edwards, 2015-02-27:
On the "shared xacro", make sure you namespace everything in such a way that it doesn't have naming collisions.  In the past I've found that multiple xacro definitions of something like `pi` have caused issues.  Unfortunately, I can't remember the "right" way.  I know....I'm no help :)

gavanderhoorn, 2015-03-17:
This would be partially solved by #23 I think, @Levi-Armstrong?

Levi-Armstrong, 2015-03-17:
@gavanderhoorn: You are correct, it does correct the rgb for the colour yellow. I believe the only open question is, if yellow is the appropriate colour for the collision geometry.

gavanderhoorn, 2015-03-18:
I guess that is up to us. MoveIt will colour meshes reddish when they are in collision, iirc, so it would only be a visual aid for when inspecting the model or visualising state I guess. For (my) other pkgs, I've always just left collision and visual geometry material colour the same, but I can see the value of having separate colours.

I've no real preference.

Levi-Armstrong, 2015-03-18:
If you have no preference I vote to leave it as is, the colour yellow.

gavanderhoorn, 2015-03-19:
@shaun-edwards: do you know why yellow was used for the ABB urdfs specifically?

gavanderhoorn, 2017-03-09:
Afaict, all xacro macros now use `abb_resources` for the colours of the meshes (as far as they don't use Collada meshes).

Closing this as it's no longer an issue.

closed 2017-03-09 by gavanderhoorn