ros-industrial/abb_experimental #29: How can I connect ABB IRB120 on the RobotStudio

james2254, 2015-12-25:
After setting parameters，i run the server program，virtual FlexPendant report errors。How can solve this problem？I follow the steps in the ros wiki。The website is http://wiki.ros.org/abb/Tutorials。My robot is ABB IRB120。
[image]

Levi-Armstrong, 2016-01-10:
@james2254: Were you able to resolve the issue; if not maybe I can help?

james2254, 2016-01-11:
Thank you；I have solved this issue。

gavanderhoorn, 2016-04-08:
@james2254: can this be closed?

closed 2016-10-11 by Levi-Armstrong

merium, 2016-11-07:
Hi 

How was this issue resolved?

gavanderhoorn, 2016-11-07:
@merium: there shouldn't really be an issue: using either RobotStudio or real hw should work.

If you have a specific issue, could I ask you to open a new one?