ros-industrial/abb_experimental #36: Socket communication between ROS and ABB controller

merium, 2016-11-08:
Hi! I'm trying to connect ABB IRB 1410 to ROS but when I run my code on the abb controller, it immediately gives me this error(in the figure below). The IP address I'm using is 192.168.125.1, 1025 (if I don't specify the port number my controller crashes). Any help with this issue would be great, thanks!

[image]

gavanderhoorn, 2016-11-08:
Hi, thanks for the report.

Could you please add some more information to your OP? What version of RobotWare are you running, how did you setup the programs, etc.

gavanderhoorn, 2016-11-08:
Btw: could you open future issues over at [ros-industrial/abb](https://github.com/ros-industrial/abb/issues)? That is the main repository for the ROS-Industrial ABB packages.

Thanks.

merium, 2016-11-09:
Hi, sure no problem I'll open issues there next time. 

Our RobotWare verison is 5.60. I followed the instructions on [http://wiki.ros.org/abb/Tutorials/InstallServer](url) page for adding the codes on the robot controller side. Initially I was getting an error posted here [https://github.com/ros-industrial/abb_experimental/issues/29] which was solved once I used the IP address 192.168.125.1 along with the port no 1025. If I don't specify the port number the controller crashes and deletes all our other modules. As soon as I PP to Main, this error pops up and I am unable to establish any connection between ROS and abb controller.

merium, 2016-12-20:
Problem solved by reinstalling everything from scratch. It still shows up sometimes but when I warm restart it goes away.

gavanderhoorn, 2017-03-09:
Seems this is resolved @merium, so closing.

Thanks for reporting back how you solved it.

closed 2017-03-09 by gavanderhoorn