ros-industrial/abb_experimental #42: update abb-drivers to ROS 2.0

AustinDeric, 2017-03-29:
Add the [abb_driver](https://github.com/ros-industrial/abb/tree/indigo-devel/abb_driver) package and migrate it to ROS 2.0.

gavanderhoorn, 2017-03-29:
Can you describe a bit more what you want to do?

`abb_driver` is already part of the main `abb` repository, and mixing Ament and Catkin packages is not a supported use-case afaik in ROS 2.

shaun-edwards, 2017-03-29:
Updating to ROS 2.0 is probably a much larger discussion.  Is there a specific reason you want to make this change.

gavanderhoorn, 2017-03-29:
> Updating to ROS 2.0 is probably a much larger discussion.

Yeah, I was just going to add that `abb_driver` depends on a lot of infrastructure, and just upgrading that in isolation seems like a strange thing to do.

AustinDeric, 2017-05-09:
too early, thanks for the discussion.

closed 2017-05-09 by AustinDeric