ros-industrial/abb_experimental #46: Transition robot models to main repository

Jmeyer1292, 2017-04-25:
The `irb120`, `irb120t`, and `irb4400` have all been in this repository for > 2 years and should be transitioned to the main abb repo.

AustinDeric, 2017-04-26:
I would like to see all the *_experimental repos be merged with the main repo.  They should still be clearly identified This would allow us to:
- save time
- focus efforts on 1 repo
- allow the experimental robots to be tested because the main repos get more use.

gavanderhoorn, 2017-04-26:
Just a data point on the 'visibility' of (support) pkgs in experimental repositories: the pkgs in `fanuc_experimental` get about as much use as those in `fanuc` (judging by the nr of questions / support requests I receive about them), which I believe is mainly due to the fact that the ROS buildfarm indexes both repositories *and* both collections have ROS wiki pages ([wiki/fanuc](http://wiki.ros.org/fanuc), [wiki/fanuc_experimental](http://wiki.ros.org/fanuc_experimental)) that also cross-reference each other.

I've recently submitted the `abb_experimental` repository for indexing by the buildfarm and created an initial wiki page for it as well ([wiki/abb_experimental](http://wiki.ros.org/abb_experimental)). Updating the `abb_experimental` metapackage to `exec_depend` on the appropriate pkgs should bring visibility of those on-par with the main repository.

gavanderhoorn, 2017-04-26:
@Jmeyer1292 wrote:
> The `irb120`, `irb120t`, and `irb4400` have all been in this repository for > 2 years and should be transitioned to the main abb repo.

Should they first be updated to follow the [ABB naming conventions](https://github.com/ros-industrial/rep/blob/master/rep-I0007.rst)?

Levi-Armstrong, 2017-04-28:
> Should they first be updated to follow the ABB naming conventions?

I agree with updating them to the latest naming convention prior to transitioning them to the main repository.

Jmeyer1292, 2017-04-28:
Well I look forward to reviewing the pull requests from the maintainers of this package!

gavanderhoorn, 2017-05-05:
I think the visibility concerns for the packages in this repository have been addressed now that the ROS doc indexer has rebuilt everything today. See [wiki/abb_experimental](http://wiki.ros.org/abb_experimental).

AustinDeric, 2017-05-26:
Back to @Jmeyer1292's original comment, do we have requirements that need to be satisfied to move these robots to the main repo?  Are there any measurable metrics?

gavanderhoorn, 2017-05-26:
> Are there any measurable metrics?

No, not at this time.

> do we have requirements that need to be satisfied to move these robots to the main repo?

yes.

The point of having things in an experimental repository is that it allows the maintainers to make any potentially breaking changes without prior notice, that it allows for some soak time for new packages and to make them available to users to expose them to use-cases that are not necessarily those of the original author / maintainer.

After a certain amount of time packages stabilise and could be moved to the main repositories. At that point, the regular rules start to apply: semantic versioning, backwards-compatibility guarantees (tick-tock, etc) and all the rest.

Tbh I don't quite get what the problem is with this scheme. If it's lower visiblity then I believe we've addressed that with the creation of [wiki/abb_experimental](http://wiki.ros.org/abb_experimental). If you take a look at the repository stats you'll see that that resulted in quite some traffic. My experience with `fanuc_experimental` is similar.

If it's the fact that experimental repositories are not released and have to be build from sources: I would say that anyone doing anything serious with ROS will have a workspace and runs Catkin once in a while anyway. With a proper `.rosinstall` and / or documentation it's peanuts to build the packages. But if we really want to we *could* create releases for the experimental repository as well.

The fact that #45, #53, #54, #56, #60, #63, #64, #65 and #67 were submitted -- and just for the few support pkgs that were contributed in the past few months -- I think shows that letting things sit here for some time is not such a bad thing.

Perhaps we could quantify what "a certain amount of time" is. That could take away some of the apparent 'arbitrary-ness' of when things do get migrated.

gavanderhoorn, 2025-04-05:
All support packages have been migrated to the main repository (ie: `ros-industrial/abb`).

closed 2025-04-05 by gavanderhoorn