ros-industrial/abb_experimental #78: Where is package controller_manager?

marooncn, 2017-09-25:
When I launch irb120_3_58_gazebo.launch in abb_irb120_gazebo  package. There comes an error:

ERROR: cannot launch node of type [controller_manager/controller_manager]: controller_manager
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/maroon/abb_experimental_ws/src
ROS path [3]=/opt/ros/kinetic/share

And I can't find the package in ROS.

gavanderhoorn, 2017-09-25:
[controller_manager](http://wiki.ros.org/controller_manager) is a package in the [ros_control](http://wiki.ros.org/ros_control) set of packages.

Can you describe how you setup your workspace and how you build it? Cloning just this repository is not enough, and is typically the reason for these kind of problems.

marooncn, 2017-09-25:
I cloned this package to my workspace/src path and used catkin_make to compile and there is something warning. So I rebuild it use wstool as [this tutorial](https://github.com/ros-industrial/abb_experimental/blob/kinetic-devel/README.md). But when I run the last command:
catkin build
to build, it warns that can't find catkin. So I go to [catkin tutorial](http://wiki.ros.org/catkin) to install catkin.
[image]
Although after this command I still can't run catkin command. But anyway, the package is built. So I direct run the demo.
In short, after I failed to compile using catkin_make command in workspace, I run the below command in workspace/src path:
wstool init .
wstool merge abb_experimental/abb_experimental.rosinstall
wstool update
mkdir build && cd build && cmake ../ && make && sudo make install

gavanderhoorn, 2017-09-25:
There is a divergence between the `readme` and the instructions [on the wiki](http://wiki.ros.org/abb_experimental). Try the instructions on the wiki, those should work.

The missing step is using `rosdep` to make sure you have all dependencies installed. It would seem like you don't have them, which is why compilation and later running failed.

gavanderhoorn, 2017-09-25:
@marooncn wrote:
> mkdir build && cd build && cmake ../ && make && sudo make install

Note: only in very rare circumstances would a `mkdir build && cd build && cmake ../ && make && sudo make install` be used with ROS packages. For almost all typical usage, please follow regular a Catkin workflow as is documented on the wiki.

marooncn, 2017-09-25:
@gavanderhoorn. thanks a lot. I will try it tomorrow，then feedback to you.

marooncn, 2017-09-26:
@gavanderhoorn I tried and it worked. But there are still some warnings when I launch irb120_3_58_gazebo.launch like the below:
[ WARN] [1506399359.754642888, 0.162000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_1'.
and I use
 '<hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>' to replace
 '<hardwareInterface>PositionJointInterface</hardwareInterface>' in  irb120_3_58_macro.xacro file
and the warnings missed.

gavanderhoorn, 2017-09-26:
That warning is somewhat expected, as the `_gazebo` pkgs have been created for Gazebo 2, while you are using a newer version.

gavanderhoorn, 2017-09-26:
Related: #35.

gavanderhoorn, 2017-09-26:
Could I please ask you to close this issue if you feel it has been resolved?

marooncn, 2017-09-26:
Of course. Thank you again. @gavanderhoorn

closed 2017-09-26 by marooncn