ros-industrial/abb_experimental #97: Inertial values for IRB 1200 models?

jonbinney, 2018-04-25:
Looks like the 120 has inertial values defined in the URDF here: https://github.com/ros-industrial/abb_experimental/blob/e7e6c119c23ab257cb545450b927ed6df3920dcc/abb_irb120_support/urdf/irb120_3_58_macro.xacro

Does anyone know how those values were determined? @bhomberg and I are trying to determine those values for the IRB1200 5_90 and 7_70 models. We've contacted ABB and they aren't willing to share those values - did someone measure them?

/cc @stwirth

shaun-edwards, 2018-04-25:
I don't know where they came from, but sometimes people use the CAD model to estimate them.  As you have found, the robot vendors do not provide this information.  I would use at your own risk.

gavanderhoorn, 2018-04-26:
These were made up / guessed / approximated using SolidWorks.

See also https://github.com/ros-industrial/kuka_experimental/issues/128#issuecomment-383111315.

gavanderhoorn, 2018-04-26:
I'm going to close this as it's not really an issue with the packages.

Feel free to keep commenting though.

closed 2018-04-26 by gavanderhoorn

jonbinney, 2018-04-26:
Thanks for the info! We'll estimate the intertias as well as we can and submit a PR. They'll be useful for doing more realistic gazebo simulation of the robot.