ros-industrial/abb_experimental #107: A bug in abb_irb1200_5_90_moveit_config/launch/moveit_planning_execution.launch

spy14414, 2019-01-10:
The code is:

  
  
  <group unless="$(arg sim)">
    <include file="$(find abb_irb1200_support)/launch/robot_interface_streaming_irb1200_5_90.launch" >
      <arg name="robot_ip" value="$(arg robot_ip)"/>
    </include>
  </group>
But the file robot_interface_streaming_irb1200_5_90.launch is not exists in abb_irb1200_support/launch.
We should replace it with the following：
 <group unless="$(arg sim)">
    <include file="$(find abb_irb1200_support)/launch/robot_interface_download_irb1200_5_90.launch" >
      <arg name="robot_ip" value="$(arg robot_ip)"/>
    </include>
  </group>

gavanderhoorn, 2019-01-10:
You're correct. Thanks for reporting.

The 7/0.7 launch file has the same problem:

https://github.com/ros-industrial/abb_experimental/blob/d6900d8b56913a10558db6ca1e0e6a68d7896690/abb_irb1200_7_70_moveit_config/launch/moveit_planning_execution.launch#L36

Could you perhaps submit a PR to fix this?

spy14414, 2019-01-22:
I rarely use this website.So I don't know what I need to do. Sorry, can you teach me or give me a tutorial?
 In addition, my English is  very poor.I don't know if you can understand what I am talking about.

gavanderhoorn, 2019-05-09:
This doesn't seem to have been addressed, so I'll re-open the issue to keep track of it.

PR #120 (2020-03-12): Fix incorrect driver launch file include in IRB 1200 moveit configs
As per subject.

Fixes #107.
closed 2020-03-12 by gavanderhoorn