ros-industrial/abb_experimental #122: How to integrate new robots?

ugurkanates, 2020-04-17:
Can you show a gudeline to how to integrate robots that arent available here like IRB 360 Flex picker for instance?

I can find some ways to turn CAD models to UDSF but they look wrong and broken.

[image]

And this only imports shape of it . Individiual parts needs to be configured seperately , but cant find any source how to proceed with it . Since I'm not mechnical engineer rather a software engineer it is hard to keep up mechanical terms so it would be lovely if you could tell it in terms of layman

gavanderhoorn, 2020-05-19:
Apologies for responding late to this. Somehow I did not get any notification about your post.

There isn't a single answer to what you're asking. Instead, there are many different ways to do this.

One would be to export the relevant parts of the CAD model as separate meshes. Then combine those manually into a `.xacro`.

Another approach could be to use the SolidWorks to URDF exporter, which can automate some of this, but comes with its own limitations and peculiarities.

A complicating factor here is that the flexpicker is a delta, which includes closed loop kinematics. That is not supported natively by URDF, so you'll have to include some modelling tricks (such as not trying to create closed loops, but using something called *mimic joints*).

Finally: personally it always helps me to see how other users have approached things. For delta robots there aren't that many examples, but you should be able to find at least some which you could use to draw inspiration from.

> I can find some ways to turn CAD models to UDSF but they look wrong and broken.

Did you mean to write *URDF* here? Or SDF?

Finally: I don't recognise the program you are using to visualise your model. Please note that we don't explicitly support anything here but the regular ROS tools like RViz and Gazebo.

gavanderhoorn, 2020-05-19:
As this does not report a problem with any of the packages here in this repository, I'm going to close the issue.

Feel free to keep commenting on it of course.

closed 2020-05-19 by gavanderhoorn