ros-industrial/abb_experimental #123: some questions for creating irb6700 155/2.85 support package

EulerPiGauss, 2020-05-19:
Hi, recently I am trying to create a irb6700 155/2.85 support package based on abb_irb6700_support. I have some questions, could you give me some help?

1. According to abb's [6700 product specification](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=2&cad=rja&uact=8&ved=2ahUKEwjRkab1gL_pAhWSHKYKHe9AC2gQFjABegQIARAB&url=https%3A%2F%2Flibrary.e.abb.com%2Fpublic%2Fe7176588ac864d36b7850fe12e6c4484%2F3HAC044265%2520PS%2520IRB%25206700-en.pdf&usg=AOvVaw02TuPKwMGfSj9qruyNPl3n), 
6700 200/2.60's max axis speed is [110, 110, 110, 190, 150, 210]degree/s. However, in irb6700_200_260_macro.xacro, velocity limits are [100, 90, 90, 170, 120, 190]. It is confusing for me.
[image]

2. I noticed that in irb6700_200_260_macro.xacro, there are cylinder_joint and pistion_joint definition. And according to [this](https://github.com/ros-industrial/abb_experimental/pull/105#discussion_r239075986), limit value seems not important. I haven't found any infomation about piston and cylinder in any document until now. So could you tell me how you get these limit information please? Is it right that I keep your values unchanged for 6700 155/2.84?

3. Although in RViz, everything looks OK, I still want to know how you verify robot's urdf. Could you give some instructions?
[image]

Thank you very much!

gavanderhoorn, 2020-05-19:
> 1\. According to abb's [6700 product [...]

According to the commit history, the variant you refer to was contributed by @marip8 in #116.

@marip8: would you happen to remember where you got the joint limits for the 200/2.6m variant? They do seem to be different from the ones in the quoted documentation.

> 2\. I noticed that in irb6700_200_260_macro.xacro, there are cylinder_joint and pistion_joint definition. And according to [...]

There are no known limits for that part of the robot geometry. The limits present are there to allow for the typical motion of these parts, without restricting them. Those limits do not come from any documentation of data sheet.

If the variant you're adding has a similar structure, you could most likely reuse the ones you found.

> 3\. Although in RViz, everything looks OK, I still want to know how you verify robot's urdf. Could you give some instructions?
>        [image]

For the visual aspect of the model, a visual inspection is used (ie: does the model look complete, do the meshes look correct, are there no obvious defects, are the mesh normals correct (ie: no inverted normals), etc).

For the kinematic structure, a good approach would be to compare the result of looking up the `base->tool0` transform with the robot in various poses and then comparing it to what either a real robot or RobotStudio tells you the pose of the unconfigured tool frame is.

Both position and orientation should match to a significant number of decimals.

Next, compare the directions of rotation to what the manual specifies. You could either do this manually, for each joint, using the `joint_state_publisher`, or by running a RobotStudio simulation of the same robot variant and then starting the driver to visualise the current state. Motions in RobotStudio should correspond to what you see in RViz.

Especially important is the J2-J3 coupling factor. But if that's incorrect you'll immediately notice it when moving the robot around in RobotStudio, as it will lead to unexpected motions of the lower arm (ie: links 3 and later).

gavanderhoorn, 2020-05-19:
> recently I am trying to create a irb6700 155/2.85 support package

please note: *variants* of robot series for which support packages already exist should be placed in the package for that series.

So in your case, the 155/2.85 could be placed as a variant in the existing `abb_irb7600_support` package.

There would be no need to create a new separate package.

gavanderhoorn, 2020-05-19:
We'll track the potentially incorrect joint limits in #124.

gavanderhoorn, 2020-05-19:
As we're tracking the potential issue with the joint limits in #124, and the rest of this post is asking for guidance instead of reporting other problems, I'm going to close the issue.

Feel free to keep commenting on it of course.

closed 2020-05-19 by gavanderhoorn

EulerPiGauss, 2020-05-19:
Thank you very much for your help!

gavanderhoorn, 2020-05-19:
No problem.

It would of course be very much appreciated if you could contribute support for the variant by submitting a pull request.