ros-industrial/abb_experimental #124: Incorrect (?) joint limits specified for IRB 6700 200/2.60 variant

gavanderhoorn, 2020-05-19:
From https://github.com/ros-industrial/abb_experimental/issues/123#issue-620679287:

> 1\. According to abb's [6700 product specification](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=2&cad=rja&uact=8&ved=2ahUKEwjRkab1gL_pAhWSHKYKHe9AC2gQFjABegQIARAB&url=https%3A%2F%2Flibrary.e.abb.com%2Fpublic%2Fe7176588ac864d36b7850fe12e6c4484%2F3HAC044265%2520PS%2520IRB%25206700-en.pdf&usg=AOvVaw02TuPKwMGfSj9qruyNPl3n), 
> 6700 200/2.60's max axis speed is [110, 110, 110, 190, 150, 210]degree/s. However, in irb6700_200_260_macro.xacro, velocity limits are [100, 90, 90, 170, 120, 190]. It is confusing for me.
> [image]
>

gavanderhoorn, 2020-05-19:
Tracking this here instead of as a part of #123.

gavanderhoorn, 2020-05-19:
My comment on this in #123:

> According to the commit history, the variant you refer to was contributed by @marip8 in #116.
> 
> @marip8: would you happen to remember where you got the joint limits for the 200/2.6m variant? They do seem to be different from the ones in the quoted documentation.

marip8, 2020-05-19:
It looks like those values match the the 235/2.65 variant which was the first type for which I made a URDF. I probably just copy-pasted that URDF to create the 200/2.6 variant and forgot to update those values. I'll submit a PR with the changes shortly

gavanderhoorn, 2020-05-19:
Thanks @marip8 :+1:

closed 2020-05-19 by gavanderhoorn