ros-industrial/abb_experimental #128: ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:

zhangzhihao2020, 2021-05-08:
when I run '' rosdep install --from-paths src/ --ignore-src --rosdistro kinetic'' ,appending the issue
```
WARNING: given --rosdistro kinetic but ROS_DISTRO is "melodic". Ignoring environment.
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
abb_irb120_gazebo: No definition of [ros_controllers] for OS version [bionic]
abb_irb1200_gazebo: No definition of [joint_trajectory_controller] for OS version [bionic]
abb_irb1200_5_90_moveit_config: No definition of [trac_ik_kinematics_plugin] for OS version [bionic]
abb_irb1200_7_70_moveit_config: No definition of [trac_ik_kinematics_plugin] for OS version [bionic]
abb_irb1600_6_12_moveit_config: No definition of [trac_ik_kinematics_plugin] for OS version [bionic]
```

gavanderhoorn, 2021-05-08:
The readme mentions this about `rosdep`:

> ```bash
> # be sure to change 'kinetic' to whichever ROS release you are using
> $ rosdep install --from-paths src/ --ignore-src --rosdistro kinetic
> ```

Did you "*change 'kinetic' to whichever ROS release you are using*"?

You appear to run Melodic, not Kinetic.

gavanderhoorn, 2021-05-08:
I don't believe this is a problem with the packages in this repository, so closing.

closed 2021-05-08 by gavanderhoorn