ros-industrial/abb_experimental #129: Dependency on abb_driver

dave992, 2021-05-26:
I am currently configuring a new robot cell with an ABB IRB1200, and am therefore using this package for the URDF of the manipulator. I plan on using the new `abb_robot_driver`, but I noticed during the build tests I ran that the packages in this repository still depend on the old `abb_driver`.

Are there any plans to migrate this to the new driver or make this repository agnostic to the driver used?

gavanderhoorn, 2021-05-26:
This is a similar issue as with RViz. We discussed that a bit in https://github.com/ros-industrial/ros_industrial_issues/issues/50.

The trade-off seems to always be: the dependency can be removed, but that will mean users will be left with 'broken' `.launch` files, unless they figure out they also need to install the driver used.

An additional complicating factor in this case is that `abb_robot_driver` basically only works if you have EGM. So we cannot standardise on it, as that's not a basic option.

dave992, 2021-05-28:
> The trade-off seems to always be: the dependency can be removed, but that will mean users will be left with 'broken' .launch files, unless they figure out they also need to install the driver used.

Yeah, I understand. It just seemed to me the `robot_interface_*` launch files seem to be more in place in their respective driver packages, but that indeed does not solve the 'broken' .launch files that will happen if this is changed now. 

> An additional complicating factor in this case is that abb_robot_driver basically only works if you have EGM. So we cannot standardise on it, as that's not a basic option.

Thats a good point.

gavanderhoorn, 2022-03-22:
As I believe the initial question raised has been answered, I'm going to close this issue.

Feel free to keep commenting on it of course.

closed 2022-03-22 by gavanderhoorn