ros-industrial/abb_experimental #134: Adding a custom End - Effector to the ABB IRB120 Robot.

JacCatania, 2022-03-22:
I have used this package to be able to use the ABB IRB120 robot on Gazebo, where I managed to install the necessary files needed and control the robot. My next step would be to add an end-effector to the ready-made robot. However, I am not sure how this can be done with the use of RVIZ, or MoveIt!, or any other program. Can somebody give me an indication of what needs to be done? Note that the end effector needs to be hot-wire cutter, meaning that this would be bracket, with a wire across the bracket itself.

Thanks,
Jacob

gavanderhoorn, 2022-03-22:
You don't use any programs for this, other than a regular text editor.

See [wiki.ros.org/Working with ROS-Industrial Robot Support Packages - Example: adding an end-effector to a robot](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages#Example:_adding_an_end-effector_to_a_robot).

gavanderhoorn, 2022-03-22:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting on it, but do take into account that for general usage questions, we typically recommend posting on fora such as ROS Answers (where this topic has been discussed a few times already).

closed 2022-03-22 by gavanderhoorn

JacCatania, 2022-03-22:
Thank you for your reply and sorry for any inconvenience.